dinyar/uGMTfirmware #127: Synchronize extrapolation logic with emulator

dinyar, 2015-03-23:
The current extrapolation logic is not completely synchronized with the emulator. 

The current plan is the following:

* Any track with `pT > 31.5 GeV` will not be extrapolated
* For the extrapolation a reduced eta value will be used. It has been agreed to remove the two LSBs.
  * Furthermore the MSB can be ignored as the coarse location of a muon track is determined by the track finder it originated from
    * The MSB is set only for muons from the endcap TF. This means that care needs to be taken in this region - a "higher" 7 bit value (e.g. 0b1111111) would be actually a lower value than e.g. 0b0000001 as the MSB would be set in the latter case but cleared in the former one.

@jlingema: Do we need the same input bit width for both the eta and phi extrapolation? 

Currently I'm assuming 1 bit more for the phi extrapolation. The output (deltaPhi and deltaEta) is 4 bits two's complement for both extrapolations. It may be more sensible to use the charge bit to determine whether the value should be added or subtracted?

dinyar, 2015-03-23:
After discussion with Joschka: For the moment leave in the charge for the input, but make output unsigned as we can decide whether to add or subtracted based on the sign bit.

dinyar, 2015-04-18:
@jlingema: Just to make sure: Is the emulator currently in sync with the firmware?

For the phi extrapolation LUT the following is now set:
- 12 bit input ( bits 7 to 2 of eta followed by bits 5 to 0 of phi)
- 3 bit output (unsigned deltaPhi)

For the eta extrapolation LUT:
- 12 bit input (same as for phi LUT)
- 4 bit output (signed deltaEta)

Do you need anything else from me?

jlingema, 2015-04-18:
The emulator is in synch with this, yes.

What I'd still need (and maybe something we need to discuss) is the scale of the output. Is it the same as phi / eta (i.e. step-size of 0.011)?

If that's the case we need to discuss as that would mean a maximum dPhi (dEta) of 0.08 which is much too small. We'd rather need something covering the scale of 0.1 ... 0.8. Is it possible to take this into account when doing the addition? (In software terms: Can we left-shift the LUT output by one before adding the phi and dPhi?)

jlingema, 2015-04-20:
When I said that we are in synch, I assumed that for the inputs the phi is a typo and you meant pT, right?

dinyar, 2015-04-20:
Ah, yes. :) Regarding the output of the LUTs: Good point, it's trivial to shift by N, but I wonder if 3 bits is sufficient. We can increase this probably, I'm not sure how much precision is needed.

jlingema, 2015-04-20:
So we stick with 3 bit for now (with the possibility of revisiting this later) and do a scale from ~0.1 to 0.8... that would mean we apply a factor of 8 in terms of the normal phi/eta scale: 
0.088 ... 0.704 with step-size of 0.088

In that case I'd argue that we would want to make the cut-off at which we assume extrapolation = 0 to be configurable, as discussed.

dinyar, 2015-04-20:
Ok, I think we're as of now synchronized (at least in intent), so I'm closing this as any inconsistency should probably be treated as a bug.

closed 2015-04-20 by dinyar

jlingema, 2015-04-21:
Just one last question: Is the order of the inputs as indicated in the (old) google-docs table:
Bits 0...5 = pT and 6...12 eta? I.e. the pT input is LSW and eta MSW?

dinyar, 2015-04-21:
(answered offline:) yes, exactly.