dinyar/uGMTfirmware #164: Allow testbenches to be run in sequence as a full test suite

dinyar, 2015-04-15:
Currently each testbench has to be started and stopped manually as they idle at the end of a run. Furthermore the result of a run is currently not machine-readable. It would be good to have a small script that runs all available testbenches in sequence and reports on success or failure.

To achieve this the following needs to be done:
- Each testbench returns to the shell
- A return code is passed back somehow. In the worst case a file could be written with the test result, but ideally an error code could be passed back to bash.

jlingema, 2015-04-15:
Wouldn't it be easier to wrap all in python, write a text file from the testbench and let the py-script figure out the rest? In case there are errors you'd probably want the text file anyway, I guess.

If I can help here, let me know.

dinyar, 2015-04-15:
True. I initially thought this may be a pain with VHDL, but I think it
might actually be quite simple to modify the testbenches in this way.

Thanks for the offer! Ideally we could use something like Jenkins or Travis
CI I guess, but so far I don't know if that's at all feasible.. ( i.e. how
much suffering it would take to make Modelsim play nicely with one of
them.. ;-) )

dinyar, 2015-04-16:
So the current version writes the results of each test into its own file in the 'results' subfolder. The following tasks remain:

- All procedures in the helper function collection don't write their output into the files, but still to the terminal (this output is suppressed in the 'runAll.sh' script).
- This output should be put into a consistent form and then be parsed and converted to a format that can be understood by Jenkins. (@jlingema, is this easily doable? I'm not sure if we should put a lot of effort into this.. if it's tedious we can skip this (and the next point) and just parse for 'N errors found' messages where `N > 1`).
- Get a Jenkins server running and configure it.

jlingema, 2015-04-17:
I think it shouldn't be hard: Let's have a look at the output together.

dinyar, 2015-04-17:
With the current set up it is possible to run Jenkins locally to perform testing automatically after each commit to github.

closed 2015-04-17 by dinyar