dinyar/uGMTfirmware #170: Needed input delay

jlingema, 2015-04-17:
I am a bit puzzled by the delay (i.e. empty frames before valid bit goes to 1) I need to add:
1) Can you remind me why exactly we need it? Is it the length of the deserialiser or something inherent from the MP7-framework?
2) Why did it change? It went from 4 (<= v0.2.20) frames to 5 (v0.2.21 and v0.2.22) and then back to 4 (v0.2.23)

(see https://twiki.cern.ch/twiki/pub/CMS/L1MicroGlobalMuonTriggerFW/ for corresponding buffer-dumps)

dinyar, 2015-04-17:
Yeah, so this is a delay that originates from somewhere before the deserializers - at least that was my understanding until now. Given that it changes depending on changes in the algorithms I'm not absolutely sure anymore..

I'm going to investigate it a bit more.

dinyar, 2015-04-22:
So, to summarize my current understanding:

We're somehow introducing a delay of 2 frames during/before deserialization (i.e. at a 40 MHz clock  rising event we 'see' the 3rd frame (when counting from 1) in the deserialization buffer as the first word).
For this reason a delay of 4 frames synchronizes us properly, but we lose a bx. Therefore I'm upgrading this to a bug..

dinyar, 2015-04-25:
The previous commit should remove the delay altogether as well as remove the unnecessary delay of 1 bx in the deserializer. It should be made more clear in the code though still..

jlingema, 2015-04-26:
Sorry, I am a bit confused: My original question was less "where does the delay come from in general" and more "why did it change between versions". 

Do I understand correctly that we don't need any delay anymore?

dinyar, 2015-04-26:
Yeah, so the reason for the delay was a stupid mistake on my part: We had
introduced a delay line before the integration test with the uGT which was
2 BX long and from which you could choose which positions to use for the
deserialization (i.e. at which "time" to pick up the words). I thought I
had set the pointer to the start of the line, but I had actually set it to
the end (or, more accurately, I was filling from the end and had set the
pointer to the beginning.. ). I've now reduced the length of the line.

Furthermore it turns out that the whole thing was 1 frame "too fast" which
I also corrected (currently with an ugly hack, but I'll commit something
better soon).

jlingema, 2015-04-28:
I'm re-opening the issue because I saw some unexpected(?) behaviour when testing the latest bit-file in the MP7:

Procedure:
- Produced test-patterns with 0 frames delay
- Loaded v0.2.26 uGMT bit-file
- Ran "many_events" test

This test resulted in quite a few sorting mismatches w.r.t. the emulator. 
I re-tested with versions 0.2.24 and 0.2.23 (with the original 4 frame delay patterns) and saw similar errors.
Then ran again the 0.2.23 with delay of 5 frames => no errors.
Prepared patterns with a delay of 1-3 frames and re-ran with 0.2.26: I observed that with a 3 frame delay, the emulator and HW responses match perfectly.

I think we should re-discuss and see if we really understand the origin of this delay...

jlingema, 2015-04-28:
For completeness sake: We just tested the same thing with the "null-algorithm" provided by the MP7 developers, where the inputs are simply delayed by 2 frames and then passed on to the output. There, we observed a total delay of the data by 5 frames, where 3 frames are hardcoded in the pattern.

That means that there is no additional delay in the framework before registering the inputs in the algorithm block.

dinyar, 2015-07-21:
After discussion with the original authors of mp7fw it was determined that a delay of 1 240 MHz tick is expected. The further delay of 1 240 MHz tick was removed during the implementation of #4. In light of this I will close this as `invalid`.

commit 07ae9ef8 (2015-04-26): Fixed #170 and adjusted testbenches accordingly.
closed 2015-07-21 by dinyar