dinyar/uGMTfirmware #194: Hitting rate limiting from Github API

dinyar, 2015-04-23:
When running several tests in close succession the Github API stops serving our scripts due to the rate limit described here: https://developer.github.com/v3/#rate-limiting

We have two options:
1. Use conditional requests: https://developer.github.com/v3/#conditional-requests
2. Increase the number of requests we are allowed to make by passing our OAuth tokens to the script (probably via an environment variable set locally?)

It seems 1. is preferable if there are no large problems there. @jlingema could you look into that?

dinyar, 2015-04-23:
One of the reasons we were hitting the rate limit is that each test script was individually updating the testfiles and LUT content files.

jlingema, 2015-04-27:
After fiddling around with this a bit, I am pretty convinced that using the conditional-request with python 2.4 will be rather involved. The headers seem to be broken: When sending, they are capitalised with the internal python function: Only the first letter of a header can be capital. Therefore, using the Is-Modified-Since would require to write something from scratch or spawn a subprocess and use curl implicitly... 

I'll have a closer look at the OAuth tokens.

jlingema, 2015-04-27:
I'll lower this to feature request after talking with Dinyar since there is currently a workaround and fixing it seems rather involved for the importance of the involved scripts.

dinyar, 2015-06-03:
I'm going to close this as we almost certainly won't need it.

closed 2015-06-03 by dinyar