dinyar/uGMTfirmware #235: The address table should be cleaned up!

jlingema, 2015-06-16:
The address table has inconsistent naming in many places. Examples are:
* Sometimes numbering is suffixed with an underscore, sometimes without
* Sometimes uGMT is capitalized, sometimes not (uGMT vs ugmt)
It would be nice if this could be made consistent throughout the address-table.

Furthermore, most comments / description fields are not helpful at all. Example:
`    <node id="mu_quad0" address="0x0" module="file://mu_quad_deserialization.xml" description="Access to registers and LUTs in a muon quad." />
`
Here something like `"Access to registers/LUTs for positive endcap receiving channels"` would be more helpful. I find this is true for most of the descriptions.

dinyar, 2015-08-05:
@jlingema I've started this, but realized that the firmware logic itself doesn't care about the calo energy sum mapping to coordinates. Could you let me know which channels we expect to correspond to which eta value?

dinyar, 2015-08-05:
Ah, forgot I was still waiting for the energy sum mapping. Reopening.

dinyar, 2016-02-27:
Resigned to @thomreis. Is it useful to have the mapping of energy sum input channels to eta value in the address table? In that case, could you let me know what the mapping is in the LUTs currently?

commit 67fdf3b2 (2015-08-05): Regenerated ipbus decoders to close #235.