dinyar/uGMTfirmware #260: Further changes in address table due to software dependency

jlingema, 2015-08-13:
Some small changes to the address table are necessary to make the integration with control software easier. To that end I will do the following changes:
* all LUT nodes will have a name to trigger the RegEx =.*mem_.*=
* all LUT nodes will get the class attribute =MuLUTNode=
* At the highest level possible, a mother-node will receive the class attribute =MuLUTManagerNode=
* All =MuLUTManagerNode=s will get a parameters tag that specifies the LUT contents

The first point necessitates a change of name in the Nodes =payload.isolation.isolation_check.abs_iso= and =...rel_iso=. 

While doing with these changes I will go ahead and clean up file-naming, changing the file-names to reflect the node ids (where applicable) to make navigating the address tables a bit easier.

jlingema, 2015-09-25:
@thomreis is also interested in this.

dinyar, 2016-01-21:
I'm not sure about the status of this. Does this still have to be done?

thomreis, 2016-01-21:
There are still some things missing that we need for SWATCH yes. We need to compare out tables.

closed 2016-02-10 by dinyar