dinyar/uGMTfirmware #317: Duplicate outputs to uGT

dinyar, 2016-01-29:
The uGT can be run as a multi-board system, consisting of a maximum of 6 AMC modules. Each of these boards needs to receive a copy of the uGMT outputs. We need to send our result on 6x 4 channels in parallel. This will also necessitate to move the channels used for intermediate results to other links.

It would probably make sense to use channels 0-23 for final muons intended for uGT while channels 24-35 will carry intermediate results.

dinyar, 2016-01-29:
@thomreis is maybe interested in this issue as it might mean changes in the unpacker?

thomreis, 2016-02-01:
In that case we do not plan to unpack 6x the same muon I guess. Since the intermediate muons are currently not unpacked no changed to the unpacker necessary for the moment.

dinyar, 2016-02-23:
This has been implemented in firmware. It would be good to adapt the testbench for the integration patterns for this though (serializer and sort and cancel patterns can remain the same). We now send the outputs to the uGT six times (i.e. links 0-3 are replicated on links 4-7, etc. until link 23) and intermediates then only on links 24-31.

thomreis, 2016-03-03:
Testpatterns have been updated with https://github.com/thomreis/uGMTScripts/commit/2b7ea5366a4fec8fbda43227c71985123318910e

commit 4dde8358 (2016-03-03): Small change to serializer testbench.
Checking all output channels now. This resolves #317.
closed 2016-03-03 by dinyar