dinyar/uGMTfirmware #323: Use eta-fine bit in cancel-out unit

dinyar, 2016-02-06:
The BMTF sends an eta-fine bit that is cleared when the eta coordinate could not be measured by the dedicated chambers but had to be determined from the muon station the muon was detected in. In this case the eta assigned is the middle-point of the concerned muon station and thus very coarse. The eta-fine bit should therefore be considered by the LUTs used by the cancel-out units for the barrel-overlap region.

dinyar, 2016-02-27:
@thomreis Could you generate a LUT content file for the barrel-overlap COU that contains an input for the eta fine bit? In principle it could be with the same dR cut, but if it's easy you could make it larger (e.g. 0.3).

thomreis, 2016-02-28:
Why do you want to make the cut value larger when the eta fine bit is set?

dinyar, 2016-02-28:
I meant when it isn't set :)

thomreis, 2016-05-11:
The emulator code is in PR https://github.com/cms-l1t-offline/cmssw/pull/315

dinyar, 2016-05-13:
This probably works (currently in the sort and cancel testbench etaFine is forced to '1' so that part of the LUT is accessed correctly), I'm not absolutely sure if we have muons with etaFine bit cleared at the moment.

Related: We should add a field for the etaFine bit in the sort and cancel testbench input file and adapt the testbench accordingly.

dinyar, 2016-05-18:
It looks like the etaFine bit is used correctly, but the test samples used seemed to have not tested behaviour when the bit is not set for one of the muons. If there is a problem we will open a new ticket.

closed 2016-05-18 by dinyar