dinyar/uGMTfirmware #329: Add possibility to disable bottom/top part of track-finders

dinyar, 2016-02-26:
For cosmic running it is convenient to deactivate the inputs of the track-finders for the top of the detector. This should be done in uGMT.

dinyar, 2016-02-29:
It's probably most flexible if individual links can be deactivated, e.g. using one register per track finder (and one bit per channel).

This probably needs to go into the event setup though -- I'm not sure how much word this would be. (@thomreis?)

thomreis, 2016-02-29:
We will have to add this to the ES parameters which will not make people happy I guess.
But then it should be fairly straight forward.

dinyar, 2016-03-03:
This still needs to be tested, but is otherwise done.

dinyar, 2016-03-07:
This doesn't seem to work fully in hardware yet..

commit b93b4a85 (2016-03-07): Another try for flexible channel masking.
This should close #329.
closed 2016-03-07 by dinyar