dinyar/uGMTfirmware #344: Delay BGo Signals before use

dinyar, 2016-04-13:
Currently we use the BGo signals from TCDS immediately, however they should be delayed by approximately the master latency when used to reset counters etc.

dinyar, 2016-04-13:
This will require additions to the uGMT cell.

dinyar, 2016-04-13:
Commit 23d1d43ce3a5477b319e790d928e93c53913d561 belongs to this. @thomreis The required BGo delay is now something closer to the master latency (~39 BX).

thomreis, 2016-04-13:
What needs to be changed in the cell?

dinyar, 2016-04-13:
We'd need to set the register 

```
  <node id="bgo_delay_reg" address="0x8000004" description="Register to set delay (in bunch crossings) for BGo signals." fwinfo="endpoint;width=0">
      <node id="bgo_delay" mask="0x3f" />
  </node>
```

To a certain value. Ideally this value would be derived from the master latency (i.e. `master latency +/- N`), but if it's a separate value that's fine too.

thomreis, 2016-04-13:
OK, so we need a new command to set this. It can take two parameters: masterLatency and offset/deviation. For masterLatency we can reuse the value that is already present in the configuration.

dinyar, 2016-04-13:
Yes, that would be perfect! Thanks!

thomreis, 2016-04-13:
For when do we need this?

dinyar, 2016-04-13:
It would be good to have it sometime next week if possible, then we have time to fix the timing. It's not incredibly urgent because I initialize the delay to the master latency at reset.

thomreis, 2016-04-13:
Which address table file is this register in?

dinyar, 2016-04-13:
(answered via side channel, here for reference:) it's in mp7_payload.xml

thomreis, 2016-04-13:
The masterLatency is currently given in frames (e.g. 235 for uGMT). I understand that you want the BGo delay in BX since otherwise the mask 0x3f would be too small as well. How should masterLatencies that are not multiples of 6 be handled?

thomreis, 2016-04-13:
I assume also that the BGo delay can never be negative.

dinyar, 2016-04-13:
Yes, right. I'd suggest flooring the result of the division. If we "undershoot" we can add one BX in the offset field.

Yes, it can never be negative. Actually it also can't be less than 5 BX, however in the firmware I protect against the situation already (i.e. if a value less than 5 is in the register I disregard it and assume the minimal delay). This shouldn't have any effect in the real world as we're always "far enough" away from TCDS to never need such a small delay.

thomreis, 2016-04-13:
The command setBGoDelay is in the trunk but not deployed at P5 yet. Will do this once the new FW is deployed as well.

dinyar, 2016-05-02:
In the current implementation we delay the BGOs vs. the data stream. We however require the BGOs to arrive early vs. the data stream. This should be possible to achieve by delaying the BC0 by ~(3564-master latency) and adjusting the bunch counter by the same value.

dinyar, 2016-05-09:
The current implementation of the delay causes synthesis to take more than a day. We should fix this.

dinyar, 2016-05-10:
Will try to use dedicated FIFOs for the BGO word and the bunch counter for each clock region (should require ~2*4*9 block RAMs).

dinyar, 2016-05-16:
Currently implemented alternative: 

- Use offset when calculating expected bunch crossing number for error counters
- Generate BC reset signal at correct time based on offset from orbit length

thomreis, 2016-05-17:
Online SW setting the BGo delay offset (command: setBGoDelayOffset) is ready to be deployed.

commit 9ef6d224 (2016-04-13): Generated IPbus decoders and small fix.
This resolves #344.
commit 108a3c00 (2016-05-25): Final fix for BGO delays.
Testbench now includes dummy counter input in order to be able to deduce the correct delays in simulation.
This resolves #344, if problems are found we'll fix them in a separate issue.
closed 2016-05-25 by dinyar