dinyar/uGMTfirmware #347: Increase cancel-out LUT size for B/OMTF

dinyar, 2016-04-18:
The current LUTs for the matching of ghosts allow a maximum dEta value of 0.162. In the case of the overlap region between BMTF and OMTF dEta for the cancel-out window should be larger.

The LUT should therefore be increased in size by one bit.

dinyar, 2016-04-22:
Going to try to increase the LUT size for all LUTs for the moment.

dinyar, 2016-04-22:
This can be synthesised without problems. The LUTs for the coordinate-based cancel-out now expect 5 bits of eta instead of formerly 4 bits. @thomreis, if it's ok for you I would try to integrate my changes to the emulator with your help on Monday and then we could try to increase the LUT size in the emulator too?

thomreis, 2016-04-25:
Yes, this can be done.

thomreis, 2016-05-02:
Included in PR https://github.com/cms-l1t-offline/cmssw/pull/246

dinyar, 2016-05-18:
This was tested and will be deployed in the next release.

closed 2016-05-18 by dinyar