dinyar/uGMTfirmware #348: Add register to store algo revision

dinyar, 2016-04-24:
It would be useful to introduce a register that contains the number of the currently used uGMT version. Ideally this should be checked by the uGMT cell (this will interest @thomreis)

dinyar, 2016-04-24:
It might be useful to take this opportunity to make the versioning scheme more explicit. I was imagining something along the following lines:

- Mayor version (first octet): Changes that require significant interventions at p5. e.g.: online software version changes, link protocol changes, etc.
- Minor version (second octet): Changes that require modifications to offline software (i.e. the emulator). i.e. algorithm changes
- "Very minor" version (third octet): Changes and additions to monitoring facilities
- Patch level (fourth octet): Bug fixes

thomreis, 2016-04-25:
I agree that this would be a useful feature to have. I am not sure if this is something that needs to be checked in the monitoring though. The number would only change on a cold reset, so we could make a command that checks during the setup step if the version is the one expected.
Of course we could do both.

dinyar, 2016-04-25:
A check at setup sounds good to me!

Cheers,
Dinyar

thomreis, 2016-04-25:
This would mean that the FW version is also stored in the configuration DB.

closed 2016-04-25 by dinyar

dinyar, 2016-04-25:
This has actually been available via the mp7fw for a while: https://github.com/dinyar/uGMTfirmware/blob/993386d5300549559eb9a316a569f2568777c87d/uGMT_algos/firmware/hdl/top_decl.vhd#L14

We should use that facility instead.

thomreis, 2016-04-26:
There is a new common MP7 SWATCH command that checks the FW versions. It just needs to be executed.

dinyar, 2016-04-26:
Ok, perfect thanks. Are you "unpacking" the hardware register word? i.e. can the version be written as e.g. 2.1.0(or 2.1.0_patch5) in the database and the cell compares it against the word 02010000 (or 02010005)?

thomreis, 2016-04-26:
I have to speak with Alessandro about that. As I understand he just compares to two numbers. One for the MP7 FW and one for the algo.

commit 60ead213 (2016-04-25): Regenerated ipbus decoder logic.
This fixes #348.