dinyar/uGMTfirmware #369: Increase range of phi extrapolation

dinyar, 2017-03-01:
Following studies done on the invariant mass resolution of muons with extrapolated coordinates the following changes should be implemented:

- Increase output of phi extrapolation LUT by 1 bit and shift this value by 1 bit less than before (3->2)
- Use 7 bit of the pT value at the input and only 6 bit of the eta value (instead of 6 bit of each value)

thomreis, 2017-03-01:
It should be only 5 bit of the eta value. total input bits stay at 12.

dinyar, 2017-03-01:
Ah right, that was a typo in the comment :) Thanks for pointing it out!

dinyar, 2017-03-08:
Apparently I forgot to implement the second bullet point. Still need to change the inputs of the extrapolation LUTs.

dinyar, 2017-03-09:
Tests pass, this seems to work.

commit 5a54b701 (2017-03-02): Increased output width of phi extrapolation LUT.
Closes #369
closed 2017-03-09 by dinyar